arduino/openwrt-yun #34: Latest code fails in ruby package

ndupreez, 2015-01-14:
When attempting to build the latest version of the code:

It failed to find the ruby package:
`--2015-01-14 16:48:54--  http://downloads.openwrt.org/sources/ruby-1.9.3-p429.tar.bz2
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-01-14 16:48:55 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[3]: *** [/home/ndupreez/wspace/wiline/openwrt-yun/dl/ruby-1.9.3-p429.tar.bz2] Error 2
make[3]: Leaving directory `/home/ndupreez/wspace/wiline/openwrt-yun/feeds/packages/lang/ruby'`

When I download it manually (from ruby-lang) and copy it into the DL_FOLDER directly, it builds, but  then fails missing some libraries:
`Package ruby-core is missing dependencies for the following libraries:
librt.so.0
libruby.so.1.9
make[3]: *** [/home/ndupreez/wspace/wiline/openwrt-yun/bin/ar71xx/packages/ruby-core_1.9.3-p429-2_ar71xx.ipk] Error 1`

I looked in the OpenWRT sources directory and that particular version isn't present anymore.

Any thoughts?

RedSnake64, 2015-02-16:
Did you try the following solution?
https://github.com/arduino/openwrt-packages-yun/pull/24

sngl, 2015-03-06:
This commit fixes the issue https://github.com/arduino/openwrt-packages-yun/commit/9617fa20e0888321e0ed262a17056c9a54ea866c

closed 2015-03-06 by sngl