arduino/openwrt-yun #35: Adding spi-gpio-custom

hallard, 2015-01-16:
Hi there,

I tried to use (kmod) spi-gpio-custom but I can't opkg it, I just saw it's not on the repo. As i2c-gpio-custom is in, would it be possible to add the same for SPI ?

thank you for your help

ffissore, 2015-01-16:
Hi
As you can read from https://randomcoderdude.wordpress.com/2013/08/15/spi-over-gpio-in-openwrt/, module kmod-spi-gpio-custom has been added in OpenWRT Barrier Breaker, whereas the yun image is based on the previous Attitude Adjustment
Please consider helping us in upgrading the distribution

Wuerfel21, 2015-05-28:
Just to let you know, it is possible to manually install it, so there shouldn't be a problem including it in the repo.