arduino/openwrt-yun #40: opkg update (host unreachable)

gandarez, 2015-06-24:
I can’t reach http://download.linino.org

Pinging download.linino.org [88.198.63.58] with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 88.198.63.58:
Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),

ffissore, 2015-06-25:
Linino has been replaced by OpenWRT-Yun on April 23rd, 2014 (more than an year ago).
See https://blog.arduino.cc/2014/04/23/upgrading-the-openwrt-yun-image-on-the-yun/
Please upgrade your yun following these instructions http://www.arduino.cc/en/Tutorial/YunSysupgrade

gandarez, 2015-06-25:
@ffissore Why they downgraded the UI interface? Linino had a beautiful Bootstraper UI.

ffissore, 2015-06-25:
We haven't downgraded anything. Bootstrap theme is an option we kept disabled in order to save disk space

gandarez, 2015-06-25:
@ffissore Is there a way to enable it after had installed? or Do I need to compile my own image in order to enable it?

ffissore, 2015-06-25:
Of course: `opkg install luci-theme-bootstrap`

gandarez, 2015-06-25:
@ffissore Thanks for answering.

gandarez, 2015-06-25:
@ffissore after installed the webpage gets timeout when loading.

http://192.168.1.60/cgi-bin/luci/webpanel/homepage

This webpage is not available
ERR_CONNECTION_TIMED_OUT

closed 2015-06-25 by gandarez