arduino/openwrt-yun #50: feeds.conf.default contains wrong URL

AcrylicShrimp, 2019-09-13:
I'm trying to build, but it seems to feeds.conf.default file contains some dead URLs.
Below URLs are not working currently:

`src-svn xwrt http:///x-wrt.googlecode.com/svn/trunk/package`
`src-svn luci https://github.com/openwrt/luci/branches/luci-0.11/`

Is this repo still active? It looks like no one manage it anymore.
If yes, anyone could let me know the latest URL?

Thank you.

c1p81, 2019-10-31:
I have tried changing the links
 
src-svn packages https://github.com/arduino/openwrt-packages-yun/trunk
src-svn xwrt https://github.com/x-wrt/x-wrt
src-svn luci https://github.com/openwrt/luci/tree/luci-0.11
src-git routing git://github.com/openwrt-routing/packages.git;for-12.09.x
src-git alljoyn https://github.com/alljoyn/core-openwrt_feed;attitude_adjustment

seems to works but the compilation fails again

nsmith1024, 2020-12-26:
Hello anybody knows the github link tot the full repository of OpenWRT for Arduino YUN for the rev-2 board so i can build that OpenWRT myself from scratch?

Thanks

KK6FPP, 2022-08-23:
I don't have all the answers on how to do a software upgrade for the Arduino-Yun (original or rev2) in 2022 but here is a list of some of the challenges:
- the binary image downloads have gone from the Arduino website (I have not been able to find alternate locations)
- the recommended build environment is based on Debian Wheezy which is end-of-life and has been archived (but at least it is still available)
- trying to install Debian Wheezy from a "netinst" image will fail because the sources have moved off the mirrors listed in the image. Using the 3-DVD set of images for the initial install works but still requires updating the /etc/apt/sources.list to search archive.debian.org or archive.kernel.org instead of the listed mirrors. For example:
"deb http://archive.debian.org/debian/ wheezy main"
"deb-src http://archive.debian.org/debian/ wheezy main"
- the security updates for Debian Wheezy cannot be accessed because they are deemed too old (may not be needed, only using the OS as platform for cross-compiling the OpenWRT-YUN software)
- the FIRST_SETUP_debian_wheezy.sh adds "mirror.switch.ch" for the wheezy-backports but that does not work. Tried this instead:
"deb http://archive.debian.org/debian/ wheezy-backports main"
- after "git clone ..." but before "build.sh" you need to copy feeds.conf.default to feeds.conf and update the feeds.conf file (as mentioned in this issue #50). Feeds "packages", "luci" and "routing" worked for me. Package xwrt no longer exists at x-wrt.googlecode.com and while there is a github.com/x-wrt it is not clear whether that is really a suitable substitute.
Package alljoyn no longer exists at git.allseenalliance.org but the correct code appears to exist at github.com/alljoyn and the feeds.conf posted by c1p81 worked for me.
- there are attempts during the build process to download additional software that fail in general for two reasons:
1.) the download is attempted over TLS/SSL (to be precise, the source URL is http but the server is configured to force a redirect to https) and the server certificate isn't recognized by the OS. Try updating the ca-root certificates on Debian Wheezy or add the --no-check-certificate option to wget.
2.) the requested URL no longer exists (such as xz-5.0.4.tar.bz2 for example, but you can take your chances with xz-5.0.5.tar.bz2 instead).
At that point it becomes tedious and it is clear that the result will not be an exact reproduction of the software that was originally released for the Arduino Yun. I may revisit this project and try to complete it but for now it feels like I'm putting too much time into a device that has been discontinued and abandoned.